balena-os/balena-raspberrypi #1236: Update Linux kernel to 6.12.23 or later to include fix for SDHC over alternate GPIO pins on Raspberry Pi 5 / CM5

HakanL, 2025-04-28:
There is a bug in the Linux kernel driver for SDHC that affects Raspberry Pi 5 and CM5 that use alternative GPIO pins to access a SD reader. It's been fixed in 6.12.23 and later versions (I assume, since it's been merged into the 6.12 branch). **I'm requesting that BalenaOS is updated to the latest kernel version**, to include this fix as it breaks hardware that try to load the official overlay with the alternative GPIO pins.
See discussion here: https://forums.balena.io/t/linux-kernel-upgrade-request/374086

klutchell, 2025-05-05:
Hey @HakanL , kernel 6.12.25 is now available in balenaOS [v6.5.24+rev1](https://github.com/balena-os/balena-raspberrypi/releases/tag/v6.5.24%2Brev1) and later for these devices.

closed 2025-05-05 by klutchell

HakanL, 2025-05-05:
> Hey [@HakanL](https://github.com/HakanL) , kernel 6.12.25 is now available in balenaOS [v6.5.24+rev1](https://github.com/balena-os/balena-raspberrypi/releases/tag/v6.5.24%2Brev1) and later for these devices.

Excellent, I can confirm it's working! Thank you!

commit 5f86811a (2025-05-05): linux-raspberrypi: Update to 6.12.25

This also unifies kernel and balena-bootloader versions.

Fixes: #1236

Changelog-entry: Update kernel to 6.12.25